kantlivelong/OctoPrint-GCodeSystemCommands #2: Use subprocess.check_output instead of system.os

eyal0, 2017-03-30:
subprocess.check_output or even subprocess.Popen would allow the output of the command to be sent to the logs, too.

http://stackoverflow.com/questions/18739239/python-how-to-get-stdout-after-running-os-system

kantlivelong, 2017-03-30:
Is the interest in capturing the output for debugging purposes or just informational during manual execution?

eyal0, 2017-04-03:
Just so that the information will be in the debug logs.  Seems like it could be useful.

For my application, I'd like to make an OCTO gcode that take a photo and sends an email.  I'm currently using config.yaml's print finished hook.  The disadvantage there is that the hook runs as soon as the print is done but before the end script is complete.  I need the end script to move the platform into position for photography.

This plugin could help me because I could simply add an OCTO gcode to take a photo and send an email and put that in the end script.  However, the hook gets the print time as an input but the OCTO gcode doesn't.  I haven't yet figured out a solution.  If you have any ideas, I'd be interested.

kantlivelong, 2017-04-03:
I'll look at putting the info in the DEBUG log. 

Best I can think of is you may be able to use the OctoPrint API to pull the info but not sure when it's made available.

eyal0, 2017-04-04:
One thing that you could do is pass some variables as arguments.

For example, OCTO10 could be `/bin/send_email.py {print_time}` and instead of:

`r = os.system(cmd_line)`

you would have

`r = os.system(cmd_line % {print_time: self._comm.getPrintTime())`

Similar to the code here: https://github.com/foosel/OctoPrint/blob/6393de8c7d42a8bbddcab7cdbb6530ea88a8c82d/src/octoprint/printer/standard.py#L1060

You could have a few variables in the future, like print time, filename, etc.  This would save the script from needing to connect to OctoPrint and authenticate, etc.  Or maybe connecting and authenticating isn't that hard?  I don't know.  I'll try.

kantlivelong, 2017-04-04:
I'll look into adding variables. Only concern is that OctoPrint may not always be running a print job and how to handle it.

kantlivelong, 2017-04-10:
STDERR/STDOUT logging available in 4f547d5dda093294f317ddb246b96d50dbb81062

closed 2017-04-10 by kantlivelong

eyal0, 2017-04-14:
Excellent, thanks!