kantlivelong/OctoPrint-GCodeSystemCommands #6: Excuse the ignorance but....

bolsoncerrado, 2017-12-19:
...would the flow be:

1) Install this plugin
2) define "Gcodes" (OCTO 10, OCTO 20,...)
3) edit (for example) END GCODE on the slicer program and add OCTO 20?

I ask because the gcode will be sent to the firmware instead, right?

Or this plugin will intercept, execute and VOID the firmware call?

kantlivelong, 2017-12-19:
Correct. The plugin will intercept the psuedo-gcode and not send it to the printer. 

This is not compatible with SD print.

bolsoncerrado, 2017-12-19:
Cool!!!
How could I launch my already created PRINTER OFF command (which triggers a WeMo switch ON/OFF) only after cooldown (aka Tool temp < 50ºC) ?
 
Any ideas?

kantlivelong, 2017-12-19:
Sounds like this would be better handled by my [PSUControl](https://github.com/kantlivelong/OctoPrint-PSUControl) plugin as it has that functionality already :smile:

bolsoncerrado, 2017-12-19:
That sounds nice, so can your plugin choose to launch a .sh instead? :)

kantlivelong, 2017-12-19:
Yep. Check the docs.

bolsoncerrado, 2017-12-19:
Awesome thanks!

closed 2017-12-19 by bolsoncerrado