kantlivelong/OctoPrint-GCodeSystemCommands #7: What is the OCTO prefix, I'm confused?

aplocher, 2018-01-06:
Hi sorry for the stupid question, but what is the OCTO prefix?  How would I make this run a command, for example, when a print job starts, or pauses, or stops, or gets an error, or (ideally) each 10% of completion (to display progress across 10 LEDs)?

Are those scenarios possible with your plugin?

Thanks!

kantlivelong, 2018-01-06:
`OCTO` would be similar to `G` or `M`.

Example `OCTO123`

They certainly are but it's up to you to determine how to implement it.

aplocher, 2018-01-06:
Oh so the .gcode file would have those lines in there (either somehow by the slicer, or manually injecting those lines myself into the file)?  And I would just define what each OCTO# code represents?

kantlivelong, 2018-01-06:
Correct. Keep in mind this does not work with SD printing.

aplocher, 2018-01-06:
Ok very cool.  I think I can figure something out.  I'm using Simplify 3D with a custom script that uploads it to octoprint via curl so it should be processed by Octoprint properly...  Now I just need to read up on how to inject a custom progress % (assuming it's possible).  Looks like there's some custom scripting that can be done upon layer change.  Thanks again!

aplocher, 2018-01-06:
Sorry one last question: can I pass arguments to the shell script?  So could I have a command: "OCTO123 10 100" then call "indicatepercent.sh 10 100" - 10 being the current layer, 100 being the total # of layers, so my shell script does the math and determines %?

kantlivelong, 2018-01-06:
Should work.

closed 2018-01-08 by aplocher