kantlivelong/OctoPrint-GCodeSystemCommands #11: Just not working

louw9669, 2018-03-05:
I created a script on my Pi that toggles a pin connected to the reset button on the controller board. the script works fine but when I try to use Gxx (xx=any number) to run /usr/local/bin/reset.sh it does nothing. I get an OK in the terminal but no reset. I'm assuming I cant use any Gcode, it would have to be unassigned, correct? what am I doing wrong?

kantlivelong, 2018-03-05:
Right now gcode parameters aren't passed but there is a PR open that I intend to merge at some point.

For now just create two gcode commands and either embed the value in the script or put it as the command line in the configuration.

louw9669, 2018-03-05:
> working (#11)
> <OctoPrint-GCodeSystemCommands@noreply.github.com>

I think I may be mistaken on how this works then. what is the OCTO value that we enter? I thought when you type G then the octo number in the terminal it runs the script thats assigned. am I right?

kantlivelong, 2018-03-06:
The number is just the suffix of the virtual GCODE command:

IE:
111 = /bin/ls

Command would be `OCTO111`

louw9669, 2018-03-06:
I got it now. I'll have to try it when I get a chance. Thanks for the info.

closed 2020-04-24 by kantlivelong