kantlivelong/OctoPrint-GCodeSystemCommands #13: Sending Gcode System Commands to powered off printer

Spyder19, 2018-06-13:
I replaced my power supply with an ATX and am using the Pi to control a relay which turns the power to the printer on and off. I wrote a simple script (turnon, turnoff) to send the signal to the proper pins (17 and 27) to control the relay. The commands work perfectly in SSH, and I made a Gcode System command to send the signals to the pins (gpio -g write 17 1 && gpio -g write 27 1, and gpio -g write 17 0 && gpio -g write 27 0)

The turnoff command works thru a remote signal by sending an OCTO100 command to Octoprint, but, when the printer is powered off, I can't send the OCTO101 to turn it back on. The only way I can remotely turn it back on is thru SSH

I can send a "connect" command to a printer that's disconnected, is there any way I can send my OCTO101 command to a printer that's powered off since the command actually goes to the Pi and not the printer ?

kantlivelong, 2018-06-13:
You cant send commands to a printer that's not connected.

Have you considered using [PSUControl](https://github.com/kantlivelong/OctoPrint-PSUControl)?

Spyder19, 2018-06-13:
Well, like I said, I want to send the command **to the PI** because the Pi is what's controlling the PSU

There's also the issue of HOW I'm sending the command. It's an experiment that we call "Jarvis" that runs in Alexa's AWS test bed. But, now that I just looked at your PSUControl, that just might work

I didn't know about it. Thank You. I will see if I can make it play nice with Jarvis and let you know

kantlivelong, 2018-06-13:
PSUControl can control the PSU from the Pi and has an API for on/off/status.

Spyder19, 2018-06-13:
Okay, I'm obviously doing something wrong, or interpreting these pins wrong. The relay is connected to a hat on the Pi, which is controlled by board pins 11 and 13, or BCM pins 17 and 27, or GPIO_GEN pins 0 and 2

The relay is connected to make a short between the green and black wire, which turns on the PSU. The Pi is connected to the purple wire, which always puts out a good solid 5V

I think I've tried all the combinations, and it's not turning off, However, it DID turn off when I restarted Octoprint after I installed it, but, I had to turn it back on via SSH

Spyder19, 2018-06-14:
Ah. System Command. Found it. Works !

Thank You !

closed 2018-06-14 by kantlivelong