kantlivelong/OctoPrint-GCodeSystemCommands #16: configuration missing after reboot

Turtletrumpet, 2020-04-27:
Hello,

many thanks for your plugin. I use your plugin to shutdown my printer after finished the print. A few days ago i upate your plugin to version 1.0.0. 
After Update the configured command are lost after reboot. I uninstall completely and install again, but the problem is the same. Sometimes the configuration ist lost without reboot?!
I use octoprint 1.4.0 on a raspi 3 with an updated octopi 0.17.
Any idea to help? Thanks.

regards,

Turtle

Turtletrumpet, 2020-04-27:
Hello,

the missing display of the configuration seems to be a browser problem. I'm not sure, but the defined OCTOXX commands are working.

Regards,

Turtle

kantlivelong, 2020-04-27:
@Turtletrumpet I'm actually am seeing this myself now. Out of curiosity; If you open settings a second time do they appear?

kantlivelong, 2020-04-27:
This should be fixed in v1.0.1

kantlivelong, 2020-04-27:
Should be fixed in v1.0.1

Turtletrumpet, 2020-04-28:
Today I get the update message to 1.0.1, but my printer is working. I'll have to wait about 10 hours ...

Turtletrumpet, 2020-04-28:
I update to 1.0.1. My attempts to make the configuration settings "invisible" were unsuccessful. Excellent! Thanks!

closed 2020-04-28 by Turtletrumpet