kantlivelong/OctoPrint-GCodeSystemCommands #17: How are optional parameters passed to the system command?

justinmbrock, 2020-04-28:
For example, I have a binary system command `/home/pi/foo` that accepts command line arguments such as `/home/pi/foo a b c`. If I configured the gcode command `OCTO3` to execute `/home/pi/foo`, would sending `OCTO3 a b c` work and pass the gcode parameters to the binary as command line arguments?

kantlivelong, 2020-04-28:
You need create a wrapper script/application that reads the environment variables and does whatever is necessary.

justinmbrock, 2020-04-28:
Thank you. I'm afraid I'm not that much of a programmer. What I have is a python script that needs root privileges to control a NeoPixel stick. Since any commands executed from within OctoPi inherit its non-root privileges I wrote a simple C++ binary to parse the command line arguments and execute the python script with PyRun_SimpleFile. I'm afraid I have no idea how to modify it to parse the environment variables. Is there somewhere you can point me with how it could work? I posted my code here, if you can take a look at it: https://github.com/justinmbrock/OctoPi-NeoPixel-Control

kantlivelong, 2020-04-28:
You can post the question on https://community.octoprint.org

closed 2020-04-28 by kantlivelong

justinmbrock, 2020-04-28:
I asked you because you wrote the plugin. "Create a wrapper to read the environment variables" doesn't help any. You have no documentation. What environment variables? What format are they given in? PyRun_SimpleFile only accepts wchar_t*, so I need to know if I need to convert it.

kantlivelong, 2020-04-28:
This isn't support. The settings tab of the plugin will tell you what the environment variable names are.