kantlivelong/OctoPrint-GCodeSystemCommands #19: wont work

00ihsan, 2020-05-10:
#### What were you doing?
i made a OCTO100 GCODE command and linked it to /home/pi/Power_on.sh
in that file i have this : curl -X POST https://someLink.com
after saving and rebooting the gcode command wont work. but when i run the script it sends a post request. so it is somewhere in this application.

1. ...
2. ...
3. ...

#### What did you expect to happen?
the smart plug to be turned on.
#### What happened instead?
nothing
#### Version of OctoPrint-GCodeSystemCommands
1.0.1

#### Operating System running OctoPrint
raspbian latest from octoprint website
OctoPrint version : 1.4.0
OctoPi version    : 0.17.0

#### Printer model & used firmware incl. version
Ender 3 Pro motherboard v1.4.4 firmware 1.1.6.2

#### Link to octoprint.log with octoprnt.plugins.gcodesystemcommands set to DEBUG
i cant set it to debug

kantlivelong, 2020-05-10:
What's preventing you from sending debug logs?

00ihsan, 2020-05-10:
I don’t know how to set it in debug mode.

kantlivelong, 2020-05-10:
https://docs.octoprint.org/en/master/bundledplugins/logging.html

00ihsan, 2020-05-10:
[octoprint.log](https://github.com/kantlivelong/OctoPrint-GCodeSystemCommands/files/4606348/octoprint.log)

kantlivelong, 2020-05-10:
Removed the log as it seemed to contain some sensitive info. After looking at the logfile supplied it seems like it is executing the commands and returning success. Cannot say if the desired outcome is achieved but seems like the plugin is working as intended.

```
2020-05-10 16:14:44,742 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO100 = curl -X POST https://REMOVED
...
2020-05-10 16:19:34,589 - octoprint.plugins.gcodesystemcommands - INFO - Executing command ID: 100
2020-05-10 16:19:35,026 - octoprint.plugins.gcodesystemcommands - INFO - Command ID 100 returned: 0
```

kantlivelong, 2020-05-10:
I'll also note that debug logging was not enabled OR nothing was run after enabling debug logging.

00ihsan, 2020-05-10:
Later in the log i make a script. Doesnt seem to run that script tho?

kantlivelong, 2020-05-10:
> Later in the log i make a script. Doesnt seem to run that script tho?

Returns success.

```
2020-05-10 16:46:16,847 - octoprint.plugins.gcodesystemcommands - INFO - Add command definition OCTO100 = /home/pi/Power_on.sh
...
2020-05-10 17:12:12,787 - octoprint.plugins.gcodesystemcommands - INFO - Executing command ID: 100
2020-05-10 17:12:13,235 - octoprint.plugins.gcodesystemcommands - INFO - Command ID 100 returned: 0
```

00ihsan, 2020-05-10:
Strange. Script works. This works. What doesn’t work?

kantlivelong, 2020-05-10:
Not sure but the discussion doesn't really belong here. Try the OctoPrint [Forums](https://community.octoprint.org/) or [Discord](https://discord.octoprint.org)

Closing.

closed 2020-05-10 by kantlivelong

00ihsan, 2020-05-10:
When i send OCTO100 in cura monitor tab, it works! Great plugin btw! But when its in the gcode it wont detect the OCTO100 command. I think octoprint sends it then to the printer.

kantlivelong, 2020-05-10:
Would need octoprint.log and serial.log showing the issue.

00ihsan, 2020-05-11:
[serial (2).log](https://github.com/kantlivelong/OctoPrint-GCodeSystemCommands/files/4607346/serial.2.log)

00ihsan, 2020-05-11:
test STL: https://we.tl/t-0vYXWHhsWZ

00ihsan, 2020-05-11:
i have found the problem. it's cura related. i have to manualy insert the gcode before the bed heads up. the start gcode runs after that.

00ihsan, 2020-05-11:
solved it. tnx for your help!