kantlivelong/OctoPrint-GCodeSystemCommands #20: Requirement to explicitly call sh to execute shell script : should be in config description

edster99, 2020-07-16:
#### What were you doing?

Wanted to execute code 

Iniital error was a 'returned 127'
resolved by adding absolute path to the shell script  /home/pi/OctoPrint/OCTO100.sh (execution properties set to all)
Then got an error 2
resolved by adding sh in the gcode system commands to give sh /home/pi/OctoPrint/OCTO100.sh
This then executed successfully.

#### Version of OctoPrint-GCodeSystemCommands 
1.0.1

#### Operating System running OctoPrint
Raspbian 10 buster
Octoprint 1.4.0
Octopi 0.17.0
Python 3.7.3

#### Printer moprodel & used firmware incl. version

#### Link to octoprint.log with octoprnt.plugins.gcodesystemcommands set to DEBUG

kantlivelong, 2020-07-16:
I'm unable to reproduce this. Need logs.

edster99, 2020-07-16:
I'll create them in the morning after 'unfixing :)'

edster99, 2020-07-17:
[octoprint.log](https://github.com/kantlivelong/OctoPrint-GCodeSystemCommands/files/4936418/octoprint.log)
This is the log where I have the path defined, but not explicitly calling the script using sh.

edster99, 2020-07-17:
[image]
This is what you get with the default creation of the command returned in the terminal.    I will withdraw my request about the full path as you have that implicitly referred to in some of the screen shots.

ruedli, 2024-04-28:
I just added a shell script to run. I did three things:
-  I placed the file in a location of my PATH envirronment variable, I used /usr/local/bin
- I made sure the firstline had a shebang line, I used #!/bin/sh
- I made the file executable: chmod +x commandfile

This worked OK, I did NOT have to give it a .sh extension (running octoprint ubuntu server, using dietpi)
I did NOT have to add a path to the command
I did not need to put /bin/sh in front of the command

Could it be one of these three things I did was forgotten by @edster99 ?

[image]

OCTO10 executed the script in /usr/local/bin/light (and switched of the light in my enclosre through homeassistant with nodeRed

[image]

So.... I cannot confirm a bug.

turkishfilms, 2024-09-20:
my OCTO10 gcode command is /usr/local/bin/co.sh   the co.sh file is: 
#!/bin/sh
echo hello
i get error 127 when i run it. 
when i run OCTO50 echo pwd i get ok. but nothing shows up in any logs
i eventually want to be able to control my Pi's PWM port 15 5kHZ so i can trun my laser off and on so the engraving letters arent connected

turkishfilms, 2024-09-20:
ive been trying for hours and i cant get it to work. So frustrating