kantlivelong/OctoPrint-GCodeSystemCommands #22: does not work when called from klipper

Goodfeat, 2021-07-17:
#### What were you doing?

1 creat script autofile.sh in /home/pi/autofile.sh
2 creat gcode system command 
[image]

3. created a special octo1 macro and tried to call it ( its work!) 
[image]

5 add command to file profile.cfg
[image]

6 trying to call a command using the Test macro
[image]

this does not work

#### this does not work

#### What happened instead?

#### Version of OctoPrint-GCodeSystemCommands

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Link to octoprint.log with octoprnt.plugins.gcodesystemcommands set to DEBUG

kantlivelong, 2021-07-17:
OctoPrint is not aware of commands executed on the controller locally. The example shown in step 6 is a response from the firmware and not an executed command.

Goodfeat, 2021-07-17:
Okay,
Is it possible to parse the response in the terminal as an action for the system command ?

Goodfeat, 2021-07-17:
The m118 command was created to be able to communicate with the server. The Klipper firmware is able to communicate with the server using the "action" commands, most likely, using the m118 command, you can also interact with the server through your plugin.

PS I also modified the M118 command in the clipper so that the clipper does not display before the message "Ecco:". but as it turned out that this will not work either

kantlivelong, 2021-07-18:
Right but that does not mean it will be interpreted as a command sent from the host. If this is something you need you can certainly modify the plugin for your needs.

Goodfeat, 2021-07-18:
Thanks, it would be nice to add a function to your plugin, but I'm afraid it will take me too long to add this function. I'm not good at python.
Please can you give a hint on the function handler?. then I'll try to think of something. But maybe I won't succeed

kantlivelong, 2021-07-18:
Using a host action command hook might be the easiest way.

https://docs.octoprint.org/en/master/plugins/hooks.html#sec-plugins-hook-comm-protocol-action

Goodfeat, 2021-07-18:
> Использование обработчика команды действия хоста может быть самым простым способом.

Ok I'll try to work with this, thanks

kantlivelong, 2021-07-18:
Good Luck. If you need any additional help check out the OctoPrint Discord or Forums

closed 2021-07-18 by kantlivelong

Goodfeat, 2021-07-18:
it seems I managed to get it to work from under the clipper firmware.
though I managed to use it only for the octo1 command (difficulties arose with regular expressions and extracting a number from a string. the number also needs to be written as a string).

At least now it works and significantly expands the functionality of the plugin.

it may not be difficult for you to modify this plugin and add measurements, correcting my mistakes.

since at the moment this is a temporary solution
[__init__.zip](https://github.com/kantlivelong/OctoPrint-GCodeSystemCommands/files/6835753/__init__.zip)

Goodfeat, 2021-07-18:
almost figured out how regular expressions work.
I found something that did not allow me to read the line correctly
The klipper outputs in before of the command output "echo: ", so I inserted any characters before the word OCTO  insert the regex ".*"

It worked
```
match = re.search(r'^.*(OCTO[0-9]+)(?:\s(.*))?$', line)
cmd_id = match.group(1)[4::]
```