kantlivelong/OctoPrint-GCodeSystemCommands #23: Trouble Gertting Plugin To Execute

wapiti59, 2021-12-05:
#### What were you doing?

I am using this plugin to execute two python scripts through shell scripts, which are designed to use a servo through an Adafruit Servohat to deploy and retract a brass brush over the print bed to clean the nozzle before each print.  This was working on a previous printer, but can not seem to get it to function moved over to here.  Note:  I have rebuilt my Octopi onto an 8 GB Pi 4B+.

I have also gone through the entire community forum and tried every permutation fix even close to my problem with no success.

------------------------------------------------------------------------------
OctoPrint version : 1.7.2
OctoPi version    : 0.18.0
------------------------------------------------------------------------------

I have validated the validity of both the .sh files and their respective .py files by running them successfully from the command line after ssh'ing into the pi.

/home/pi/deploy.py
import time
from adafruit_servokit import ServoKit
kit = ServoKit(channels=16)
kit.servo[0].angle = 110

/home/pi/retract.py
import time
time.sleep(25) # delay retract 25 seconds
from adafruit_servokit import ServoKit
kit = ServoKit(channels=16)
kit.servo[0].angle = 0

/home/pi/deploy.sh
#!/bin/bash
sudo python3 /home/pi/deploy.py

/home/pi/retract.sh
#!/bin/bash
sudo python3 /home/pi/retract.py

OCTO Command Lines
OCO10 /home/pi/deploy.sh
OCTO20 /home/pi/retract.sh

NOTE: sudoers file aliases ezpi (user logged into octoprint with) to %admin, root

#### What did you expect to happen?
 After the extruder reset to the wipe position (back left corner of the bed) and the bed dropped 10mm to clear the brush, I expected the brush to deploy and then after a 24 second time delay, to then retract before it went on to auto-level and print.

#### What happened instead?
Extruder went to wipe position, went through wipe motions and then on to auto-level pre-print, brush did not deploy/retract.

#### Version of OctoPrint-GCodeSystemCommands

1.0.1

#### Operating System running OctoPrint

Pi Raspbian - 5.1.63-V71+

#### Printer model & used firmware incl. version
Latest Ender 5 Plus, Firmware as shipped two months ago.

#### Link to octoprint.log with octoprnt.plugins.gcodesystemcommands set to DEBUG

Forgive me, I am unsure how to link to my octoprint.log file, or how to put the plugin into debug mode.  I have pasted ONLY the operative lines from my log here:
2021-12-05 10:31:13,506 - werkzeug - INFO - 127.0.0.1 - - [05/Dec/2021 10:31:13] "GET /?action=snapshot HTTP/1.1" 200 -
2021-12-05 10:31:18,518 - octoprint.plugins.detailedprogress - INFO - Message: ETA 12/05 10.47AM
2021-12-05 10:31:18,907 - octoprint.plugins.gcodesystemcommands - INFO - Executing command ID: 10
2021-12-05 10:31:18,991 - octoprint.plugins.gcodesystemcommands - INFO - Command ID 10 returned: 1
2021-12-05 10:31:19,063 - octoprint.plugins.gcodesystemcommands - INFO - Executing command ID: 20
2021-12-05 10:31:19,138 - octoprint.plugins.gcodesystemcommands - INFO - Command ID 20 returned: 1
2021-12-05 10:31:19,206 - werkzeug - INFO - 127.0.0.1 - - [05/Dec/2021 10:31:19] "GET /?action=snapshot HTTP/1.1" 200 -
2021-12-05 10:31:19,703 - octoprint.plugins.octoeverywhere - INFO - NotificationsHandler successfully sent 'progress'; ETA: 911
2021-12-05 10:31:24,104 - werkzeug - INFO - 127.0.0.1 - - [05/Dec/2021 10:31:24] "GET /?action=snapshot HTTP/1.1" 200 -
2021-12-05 10:31:28,520 - octoprint.plugins.detailedprogress - INFO - Message: Good accuracy
2021-12-05 10:31:34,347 - werkzeug - INFO - 127.0.0.1 - - [05/Dec/2021 10:31:34] "GET /?action=snapshot HTTP/1.1" 200 -
2021-12-05 10:31:38,523 - octoprint.plugins.detailedprogress - INFO - Message: 7.01% complete
2021-12-05 10:31:41,767 - octoprint.util.comm - INFO - Cancelling job on behalf of user ezpi

kantlivelong, 2021-12-08:
What does `sudo python3 /home/pi/retract.py` do via SSH?

wapiti59, 2021-12-08:
Well, problem 1, it's asking for the sudo password, so that says it's not reading the suders file, OR I screwed it up somehow.  However, placing sudo at the front of the command line was one of the last things I did as well.  In the prior instance where it was working on the pi III B, it was running at the user level and working, which I duplicated when I first placed it here on the 4 B.  I can tell you running either of the shell commands, or running the python script from the command line as either the pi user, or through sudo works - they run, the brush deploys/retracts exactly as intended.  I just had a thought...  I am logging into the Octoprint instance as ezpi, but everything was created and lives under the pi user.  This is an instance created/procured through EzPi (TH3D), but there is no home directory for the ezpi user, and the only way to log into it, except of course at the console, is through ssh and only the pi user is ssh enabled.  So, tomorrow as soon as the print that's running finishes, I'll log into the console as ezpi and try to run it and see what happens, then let you know here.

kantlivelong, 2021-12-08:
Since sudo is clearly the blocker here I think we can be certain it's not the plugin. Feel free to continue on the [OctoPrint Community Forums](https://community.octoprint.org)

kantlivelong, 2021-12-08:
So the script executes succeasfully without sudo in SSH?

wapiti59, 2021-12-08:
Yes, as either user or sudo

wapiti59, 2021-12-08:
from the command line.  However, since the Pi 4 B,  it does not pass through the plugin

kantlivelong, 2021-12-08:
So if you comment out the entire py script and then uncomment one line at a time starting from the top, at what point does it break?

wapiti59, 2021-12-08:
Will have to do that this evening, but confused.  Apologies, If the py script is functioning exactly as expected from the CL, and the shell script properly calls and executes the py script, what will this tell us?  The py script is uber simple and commenting any line, other than the time delay statement in retract.py will break it.

kantlivelong, 2021-12-08:
Were just trying to see at what point the issue is occurring. Just because it works in the CLI does not always mean it will execute the same when called from another process. Environment variables can vary. Since you are getting a return code of 1 something is happening and erroring out but what?

That said, there's no indication of a bug in this plugin at this point. I've seen a few tickets/support threads with similar issues and its uaually:

- Permissions / ownership
- Using sudo and having it hang waiting for credentials
- Using relative paths
- A different python interpreter version being used resulting in missing modules.
- Script itself

Given the list above it would be good to know the permissions on the sh and py files while were testing out the py script line by line.

wapiti59, 2021-12-08:
OK.  Print Finished, running through a few things here (sorry for length):
 ls -al deploy.sh
-rwxr-xr-x 1 pi pi 40 Dec  8 14:53 deploy.sh
ls -al retract.sh
-rwxr-xr-x 1 pi pi 41 Dec  8 14:54 retract.sh
 ls -al deploy.py
-rw-r--r-- 1 pi pi 105 Nov 21 08:11 deploy.py
 ls -al retract.py
-rw-r--r-- 1 pi pi 145 Dec  5 10:07 retract.py

This is the starting script that gets built into the GCode for each print from Simplify3D (note on print initiation the printer firmware automatically homes all axies, hence commented out):
; NOZZLE CLEANING ROUTINE
; G28 ; home all axes
G1 Z20 ; lower the bed to clear the cleaning brush
G4 P2000
G1 X0 ; set the extruder to zero
G1 Y350 ; set Y to zero
OCTO10 ; deploy the cleaning brush
G1 X100 ; first cleaning pass
G1 X0 ; second cleaning pass
G1 X100 ; third cleaning pass
G1 X0 ; fourth cleaning pass and reset
OCTO20 ; retract the cleaning brush
G28 ; home all axes pre-level

; PRE-PRINT AUTO LEVEL
G29 ; Initiate Auto Level

then print from here.
I removed sudo from the equation, the shell scripts again only call python as a user.  The shells execute properly both from the console directly, as well as ssh'd into the Pi as the user pi.  NOTE:  in both places, I am logged in as pi.  I can not seem to be able to log into the octoprint UI as anything but ezpi, and I can't log into anywhere as the ezpi user.  with Sudo pulled out, I reran the routine using a test cube print with the plugin active as a base line.  Relevant output below:
2021-12-08 15:14:52,746 - werkzeug - INFO - 127.0.0.1 - - [08/Dec/2021 15:14:52] "GET /?action=snapshot HTTP/1.1" 200 -
2021-12-08 15:15:02,367 - octoprint.plugins.gcodesystemcommands - INFO - Executing command ID: 10
2021-12-08 15:15:02,574 - octoprint.plugins.detailedprogress - INFO - Message: Good accuracy
2021-12-08 15:15:02,599 - octoprint.plugins.gcodesystemcommands - INFO - Command ID 10 returned: 1
2021-12-08 15:15:02,701 - octoprint.plugins.gcodesystemcommands - INFO - Executing command ID: 20
2021-12-08 15:15:03,233 - werkzeug - INFO - 127.0.0.1 - - [08/Dec/2021 15:15:03] "GET /?action=snapshot HTTP/1.1" 200 -
2021-12-08 15:15:06,222 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2021-12-08 15:15:10,219 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2021-12-08 15:15:12,578 - octoprint.plugins.detailedprogress - INFO - Message: 6.98% complete
2021-12-08 15:15:13,706 - werkzeug - INFO - 127.0.0.1 - - [08/Dec/2021 15:15:13] "GET /?action=snapshot HTTP/1.1" 200 -
2021-12-08 15:15:14,219 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2021-12-08 15:15:18,218 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2021-12-08 15:15:22,217 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2021-12-08 15:15:22,580 - octoprint.plugins.detailedprogress - INFO - Message: ETL 00h15m15s
2021-12-08 15:15:24,199 - werkzeug - INFO - 127.0.0.1 - - [08/Dec/2021 15:15:24] "GET /?action=snapshot HTTP/1.1" 200 -
2021-12-08 15:15:26,216 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2021-12-08 15:15:27,804 - octoprint.plugins.gcodesystemcommands - INFO - Command ID 20 returned: 1

Commented out all lines in both py scripts and restarted same print.  Relevant log result:
2021-12-08 15:27:32,960 - werkzeug - INFO - 127.0.0.1 - - [08/Dec/2021 15:27:32] "GET /?action=snapshot HTTP/1.1" 200 -
2021-12-08 15:27:37,500 - octoprint.plugins.gcodesystemcommands - INFO - Executing command ID: 10
2021-12-08 15:27:37,623 - octoprint.plugins.gcodesystemcommands - INFO - Command ID 10 returned: 0
2021-12-08 15:27:37,691 - octoprint.plugins.gcodesystemcommands - INFO - Executing command ID: 20
2021-12-08 15:27:37,767 - octoprint.plugins.gcodesystemcommands - INFO - Command ID 20 returned: 0
Uncommented line 1 in each (import time)  Result:  both Command ID 10 & 20 returned: 0
Uncommented line 2 in each  Result:  both Command ID 10 returned: 1, Command 20 returned: 0 (time delay statement executes successfully)
Uncommented next line in both  Result:  both Command ID 10 and Command ID 20 returned: 1 so both fail at the import statement pulling in ServoKit, which is part of the Circuit Builder library for Python.  Still has me scratching my head here why I can run it form the command line as either a user or sudo, but not call it from within this plug-in?

kantlivelong, 2021-12-08:
`python3 --version` returns?

`ls -la /usr/bin/python*` returns?

wapiti59, 2021-12-08:
Python3 --version returns Python 3.7.3
pi@octopi:~ $ ls -la /usr/bin/python*
lrwxrwxrwx 1 root root      24 Nov 23 07:14 /usr/bin/python -> /etc/alternatives/python
lrwxrwxrwx 1 root root       9 Mar  4  2019 /usr/bin/python2 -> python2.7
-rwxr-xr-x 1 root root 2984816 Oct 10  2019 /usr/bin/python2.7
lrwxrwxrwx 1 root root       9 Mar 26  2019 /usr/bin/python3 -> python3.7
-rwxr-xr-x 2 root root 4275580 Jan 22  2021 /usr/bin/python3.7
lrwxrwxrwx 1 root root      36 Jan 22  2021 /usr/bin/python3.7-config -> arm-linux-gnueabihf-python3.7-config
-rwxr-xr-x 2 root root 4275580 Jan 22  2021 /usr/bin/python3.7m
lrwxrwxrwx 1 root root      37 Jan 22  2021 /usr/bin/python3.7m-config -> arm-linux-gnueabihf-python3.7m-config
lrwxrwxrwx 1 root root      16 Mar 26  2019 /usr/bin/python3-config -> python3.7-config
lrwxrwxrwx 1 root root      10 Mar 26  2019 /usr/bin/python3m -> python3.7m
lrwxrwxrwx 1 root root      17 Mar 26  2019 /usr/bin/python3m-config -> python3.7m-config

kantlivelong, 2021-12-09:
Good so only one version of python3 that we can see. So then the question is why is your module path list different.

You can run `python3 -c 'import sys; print(sys.path)'` to see what you get there then

```
import sys
print(sys.path)
```
in one of the existing py scripts then redirect the output in the sh script so its written to a file.

wapiti59, 2021-12-09:
Sorry - missed you - do you mean place that output ( ['', '/usr/lib/python37.zip', '/usr/lib/python3.7', '/usr/lib/python3.7/lib-dynload', '/usr/local/lib/python3.7/dist-packages', '/usr/lib/python3/dist-packages'] ) into the top of the two scripts (deploy.sh and retract.sh), or just import sys maybe?

kantlivelong, 2021-12-09:
in `test.py`
```
import sys
print(sys.path)
```

in `test.sh`
```
#!/bin/bash
python3 /home/pi/test.py > /home/pi/test.log 2>&1
```

`chmod +x test.sh`

Call `test.sh` the same way as the others using an OCTO command. Update the sh to use sudo for python3 as well and call again.

wapiti59, 2021-12-09:
thank you for the clarification.  Will run this afternoon - and THANK YOU, for all of your help sir!

wapiti59, 2021-12-09:
test,py and test.sh created as written.  Set to both OCTO15 and OCTO25

**First run as user (no sudo)**
octoprint.log:
command ID 15 & 25 returned : 0
pi@octopi:~ $ ls -al test.log
-rw-r--r-- 1 pi pi 112 Dec  9 16:40 test.log
test.log content:
['/home/pi', '/lib/python37.zip', '/lib/python3.7', '/lib/python3.7/lib-dynload', '/lib/python3/dist-packages']
**second run as sudo:**
octoprint.log:
command ID 15 & 25 returned : 1
pi@octopi:~ $ ls -al test.log
-rw-r--r-- 1 pi pi 54 Dec  9 16:51 test.log
test.log content:
sudo: no tty present and no askpass program specified

kantlivelong, 2021-12-10:
Well that's certainly interesting when comparing to the first run you did which had:
```
['/home/pi', '/lib/python37.zip', '/lib/python3.7', '/lib/python3.7/lib-dynload', '/lib/python3/dist-packages']
```

```
['', '/usr/lib/python37.zip', '/usr/lib/python3.7', '/usr/lib/python3.7/lib-dynload', '/usr/local/lib/python3.7/dist-packages', '/usr/lib/python3/dist-packages']
```

> sudo: no tty present and no askpass program specified

Right, you need to add a sudo rule to make that run UNLESS you don't intend on using sudo for your actual scripts. 

So where is the adafruit_servokit lib installed then...?
`find / -xdev -iname "*servokit*"`

wapiti59, 2021-12-10:
Every thing came back permission denied, EXCEPT:
/usr/local/lib/python3.7/dist-packages/adafruit_servokit.py
/usr/local/lib/python3.7/dist-packages/adafruit_circuitpython_servokit-1.3.6.dist-info
/usr/local/lib/python3.7/dist-packages/__pycache__/adafruit_servokit.cpython-37.pyc

wapiti59, 2021-12-10:
SO, since at the point it's trying to import that module is where it blows up, then likely it's the culprit.  So, perhaps in the top of the shell:
export PATH="$/usr/local/lib/python3.7/dist-packages$PATH" , or have I screwed up the syntax?

kantlivelong, 2021-12-10:
In the shell script you can do
export PYTHONPATH="${PYTHONPATH}:/usr/local/lib/python3.7/dist-packages"

wapiti59, 2021-12-10:
Inserted right before the python call.  BINGO!!!!  THANK YOU SIR!!!  I know you went above and beyond the call here and I can't even begin to tell you how deeply I appreciate it!

closed 2021-12-10 by wapiti59